paulober/MicroPico #249: micropico stopped working for me recently and gives `Exception occured` on code execution

anandijain, 2024-09-11:
Firstly, thank you so much for Micropico. It is about 1000x better than using Thonny. 
Up until maybe last night (9/10/24) upico was working awesome. But I booted vscode last night and the handy bar at the bottom of vscode was gone. 
[image]

I was able to get it to show up by using `Micropico: Connect` in the command palette. 
But running any code just gives me: 
[image]

I have verified that I'm able to get my esp32's LED to turn on with the following code in Thonny
```py
from machine import Pin

led = Pin(2, Pin.OUT)
led.value(1)
```

## Which version of MicroPico are you using?
v4.0.2

## Support info
```
Version: 1.93.0 (user setup)
Commit: 4849ca9bdf9666755eb463db297b69e5385090e3
Date: 2024-09-04T13:02:38.431Z
Electron: 30.4.0
ElectronBuildId: 10073054
Chromium: 124.0.6367.243
Node.js: 20.15.1
V8: 12.4.254.20-electron.0
OS: Windows_NT x64 10.0.22631
```

paulober, 2024-09-12:
HI, i'm sorry to here that the new version cause issues on for you. I would be great if you can reproduce this issue while having the "Developer console" open. You can find it in "Help" > "Developer Tools". This should hopefully give us some feedback about what is happening here.

Are you working with MicroPico in a Pico project (so initialized with MicroPico > Configure Project)?

paulober, 2024-09-12:
What Python environment have you selected with the `Python: Select Interpreter` command?

anandijain, 2024-09-14:
weird, I can no longer reproduce the issue. it's been working great all day.

I'm very sorry for the confusion.

thanks again for this extension

closed 2024-09-14 by anandijain