paulober/MicroPico #250: won’t connect unless autoConnect=False and /dev/ttyACM0 listed in manual com device when using vs code remote tunnel

jsneer23, 2024-09-11:
## What are the steps to reproduce this issue?

1. setup remote tunnel in vs code running on raspberry pi
2. connect to workspace through browser link on iPad
3. try connecting to pico board

## What happens?

a dialogue box appears in lower right corner that says no COM

## What were you expecting to happen?

i expected to connect to rp2040 device

i can connect when autoConnect is unchecked and /dev/ttyACM0 is listed in manual com device box in settings

## Any logs, error output, etc?

this action produces no logs

## Which version of MicroPico are you using?

v4.0.2

## Any other comments?

i can connect to the pico fine from the raspberry pi itself without the autoConnect workaround. this is specifically a remote tunnel issue.

## Support info

```
version:%201.93.0%0A
Commit%3A%204849ca9bdf9666755eb463db297b69e5385090e3%0A
Date%3A%202024-09-04T20%3A18%3A06.358Z%0A
Browser%3A%20Mozilla/5.0%20(Macintosh;%20Intel%20Mac%20OS%20X%2010_15_7)%20AppleWebKit/605.1.15%20(KHTML,%20like%20Gecko)%20Version/17.6%20Safari/605.1.15
```

jsneer23, 2024-09-12:
not supporting remote is not a big deal since it does work, just not seamlessly. i was on the fence about listing this as a feature request to begin with. feel free to close this as not supported if you wish.

in any event, when i type lsusb into the vs code instance on my ipad i get:
```
my_name@raspberrypi:~ $ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 2e8a:105b MicroPython Board in FS mode
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

unfortunately, at least as far as i can tell, developer tools will not work on web versions of vs code. i can try on a different device later today.

paulober, 2024-09-12:
Thanks no problem. This is all what's needed to see where the issue lies.
Could it be that you have a Pimoroni Yukon?
The reason why auto-connect doesn't work for you, is because it only filters for known ids and this product isn't currently added. 
The manualCom setting works because it just tells the extension it want's to use this exact device.
So auto-connect with this board should also not work if you are not using your tunnel setup. Can you confirm?

jsneer23, 2024-09-12:
yes i do have a yukon, which i almost stated but i had assumed it wasn’t a board specific issue because the problem was presenting only in the remote tunnel. 

namely, it does connect to the yukon automatically when using vs code directly on the raspberry pi 5 but not in the remote tunnel connected to the very same raspberry pi. ¯\_(ツ)_/¯

sorry if that wasted any of your time. thanks for being so responsive!!

paulober, 2024-09-13:
@jsneer23 No, problem I want to fix as much bugs from MicroPico as possible, so I'm thankful if some can do some debugging steps as I'm unable to reproduce most of this issues.

So that's strange, it shouldn't auto-connect in both scenarios. Can you look at the output in the Output panel near the terminal and select MicroPico in the dropdown and also "Extension Host" and see if there are any logs?
Also if you have nodejs installed, it could be very helpful if you run `npx @serialport/list` in  both scenarios.

But no matter what the results are I'll definitely will add the Pimoroni board product id to the extension so it should work in both. Though there is still the issue why it does connect in one of the senarios...

jsneer23, 2024-09-13:
ok nevermind. it turns out the pi was looking for /dev/ttyACM0 after all (manual com device was set in the folder tab of the micropico settings menu and i was only looking at the workspace tab).

so i can confirm the pi will not connect to the yukon without /dev/ttyACM0 and this is not a remote tunnel issue.

thanks for your help!!

closed 2024-09-13 by jsneer23