paulober/MicroPico #256: Linux Auto Connect does not connect, but manual connect works

sk8board, 2024-10-21:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Auto Connect provides a COM error when connecting a Pico to Linux Mint 21.3
2. Pico will not connect when using Auto Connect
3. Manual Connect works as expected

## What happens?
Receive a COM error

[image]

## What were you expecting to happen?
Pico to connect to MicroPico

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
No error shown in terminal since Pico will not connect

## Any other comments?

## Which version of MicroPico are you using?
v4.0.6

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.94.2
Release: 24286
Commit: 62f778783c52510c94e687de293bc2ad230f9a67
Date: 2024-10-12T20:14:54.465Z
Electron: 30.5.1
ElectronBuildId: undefined
Chromium: 124.0.6367.243
Node.js: 20.16.0
V8: 12.4.254.20-electron.0
OS: Linux x64 5.15.0-124-generic
```

aleb, 2024-11-20:
I can reproduce despite the serial ports being accessible:
```
% sudo chmod a+rw /dev/ttyS*
% ls -l /dev/ttyS*
crw-rw-rw- 1 root uucp 4, 64 20. Nov 16:39 /dev/ttyS0
[...]
crw-rw-rw- 1 root uucp 4, 95 20. Nov 15:05 /dev/ttyS31
```

Enabling manual connection mode led "Pico Connected":
- Micropico: Manual Com Device: `/dev/ttyS0` 
- Micropico: Auto Connect: disabled

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around connecting and reconnecting boards. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.