paulober/MicroPico #257: micropico.pyIgnore not working correctly, uploads ignored directory.

Hristo-Nikodimov-Nenkov, 2024-11-06:
> I've added .freeze directory to the micropico.pyIgnore list, however this directory is uploaded to the board (Pico W) and other ignored directories and files .micropico .vscode are not.

## What are the steps to reproduce this issue?
1. Configure project
2. Create .freeze directory
3. Edit VS Code settings:
{
    "python.linting.enabled": true,
    "python.languageServer": "Pylance",
    "python.analysis.typeCheckingMode": "basic",
    "python.analysis.diagnosticSeverityOverrides": {
        "reportMissingModuleSource": "none"
    },
    "python.analysis.extraPaths": [
        "~/.micropico-stubs/included",
        "./.freeze",
        "./lib"
    ],
    "python.analysis.typeshedPaths": [
        "~/.micropico-stubs/included"
    ],
    "micropico.syncFolder": "",
    "micropico.openOnStart": true,
    "micropico.pyIgnore":[
        "**/project.pico-go",
        "**/.picowgo",
        "**/.micropico",
        "**/.vscode",
        "**/.gitignore",
        "**/.git",
        "**/.DS_Store",
        "**/.idea",
        "**/.freeze",
    ]
}
4. Put some files in .freeze directory
5. Upload file to board using "Upload project to Pico"

## What happens?
Directory .freeze is uploaded to the board.

## What were you expecting to happen?
To upload everithing else of ignored files and directories.

## Which version of MicroPico are you using?
Latest
[image]
[image]

Hristo-Nikodimov-Nenkov, 2024-11-10:
I was able to make it work correctly by setting "**micropico.syncAllFileTypes": true**. I have no idea why it did not work without it.

Hristo-Nikodimov-Nenkov, 2024-11-14:
It also does it when I use "Toggle Mpy FS"
When I open the project directory It works.
 Then when I open the Mpy FS it si converted to WORKSPACE and starts to upload ignored directories.

paulober, 2024-11-17:
Sorry for the delay and thanks for reporting this issue.
I think the first case should work normally if you reopen vscode after setting the ignore folder as workaround. (will be fixed in next release).
And the second case should also be fixed with the next patch.

closed 2024-11-18 by paulober

Hristo-Nikodimov-Nenkov, 2024-11-20:
When project is open in workspace ( Toggle Mpy FS ) it uploads ignored files and directories.
I'm using version 4.1.0
[image]

commit 9bc9b2e3 (2024-11-17): Fix #257, Switch to machine-overridable scope for all settings