paulober/MicroPico #259: Attempting to "Upload Project", where firmware is only a subfolder of a workspace causes entire workspace to be uploaded

DeflateAwning, 2024-11-15:
Attempting to "Upload Project", where firmware is only a subfolder of a workspace causes entire workspace to be uploaded

## What are the steps to reproduce this issue?
1. Open a VS Code workspace.
2. In the root of the workspace, create a subfolder called "firmware".
3. In the root of the workspace, create a subfolder called "not_firmware". Put some random Python files in it.
4. Create a `boot.py` and a `main.py` in the firmware folder.
5. Move the `.micropico` file into the `firmware` folder.
6. Use Ctrl+Shift+P to run "MicroPico: Upload project to Pico".

## What happens?

7. See that the files in "not_firmware" also get uploaded.

## Which version of MicroPico are you using?

Default version as of 2024-11-15.

paulober, 2024-11-17:
If you want to keep your firmware in a subfolder I recommend setting the `micropico.syncFolder` setting to your subfolder containing the source. See [README](https://github.com/paulober/MicroPico?tab=readme-ov-file#extension-settings) for more details.

But it may be a good idea to detect if the .micropico file is in a subfolder and if `micropico.syncFolder` setting is empty. If that's the case, the extension could automatically adjust the syncFolder setting for you....
What do you think of this solution?

paulober, 2024-11-18:
Included in v4.1.0

closed 2024-11-18 by paulober

DeflateAwning, 2024-11-18:
Hmm setting it in a VS Code config setting seems a little strange, but I guess that's workable if it's a Workspace config setting.