paulober/MicroPico #260: The "MicroPico: Configure MicroPico Project" button doesn't open configuration

DeflateAwning, 2024-11-15:
The "MicroPico: Configure MicroPico Project" button (in the Ctrl+Shift+P menu) doesn't open configuration

## What are the steps to reproduce this issue?
1. Open a project.
2. Ctrl+Shift+P, then run "MicroPico: Configure MicroPico Project".

## What happens?

The Extensions pane opens, showing recommendations.

## What were you expecting to happen?

A configuration should open.

paulober, 2024-11-17:
I should not open any configuration. This command does configure the currently open folder as MicroPico project. The recommendations open because after configuration VS Code detects that not all recommended extensions are installed and therefore shows you the options. If you have any recommendations on how to rename this command to avoid future miss-conceptions please let me know.

closed 2024-11-17 by paulober

DeflateAwning, 2024-11-17:
"Setup workspace as MicroPico project and install extensions" ?

paulober, 2024-11-17:
Good idea but to long for the context menu. But the short form  `Setup MicroPico project` would be possible?

DeflateAwning, 2024-11-18:
I actually don't think it's too long. Accurate is better than misleading and brief.

If it's too long to explain what the button does, maybe it should be two separate buttons? No button should do magic without first explaining it to the user.

DeflateAwning, 2024-11-19:
"Initialize" is the common word for "Create tracking/config/logging files from scratch" (e.g., `git init`)