paulober/MicroPico #264: RP2040 NeoPixel `__setitem__` is missing

DeflateAwning, 2024-11-21:
The NeoPixel example from the [MicroPython rp2 docs](https://docs.micropython.org/en/latest/rp2/quickref.html) is as follows:

```python
from machine import Pin
from neopixel import NeoPixel

pin = Pin(0, Pin.OUT)   # set GPIO0 to output to drive NeoPixels
np = NeoPixel(pin, 8)   # create NeoPixel driver on GPIO0 for 8 pixels
np[0] = (255, 255, 255) # set the first pixel to white
np.write()              # write data to all pixels
r, g, b = np[0]         # get first pixel colour
```

The example gives a warning on the `np[0] = (255, ...)` line, because it can't find the `__setitem__` method of the `NeoPixel` class.

DeflateAwning, 2024-11-25:
Ah oops, I totally opened that on the wrong repo; my bad. Thanks a lot!

closed 2024-11-25 by DeflateAwning

maxi07, 2024-12-23:
Just found this issue, hope this will get updated soon!

Josverl, 2024-12-25:
Please add your vote on the micropython-stub* repo to give me better visibility on these and other improvements on stubs.