paulober/MicroPico #265: Unicode characters in REPL output

Odud, 2024-12-18:
I'm using the Micropico extension in vscode. Extended characters (unicode?) don't print correctly in the Micropico extension but work ok in Thonny. The vscode settings for terminal font family don't seem to have any effect so I suspect that the Micropico extension is using its own font family or doing something odd with multibyte chars and I can't see in the settings where this is set or how to correct it.

In Thonny:

```
MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> print('\u0420\u043e\u0441\u0441\u0438\u044f')
Россия
```

In vscode/micropico:

```
MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico W with RP2040
Type "help()" for more information or .help for custom vREPL commands.
>>> print('\u0420\u043e\u0441\u0441\u0438\u044f')
������������
```

MicroPico 4.1.0

Version: 1.96.0 (user setup)
Commit: 138f619c86f1199955d53b4166bef66ef252935c
Date: 2024-12-11T02:29:09.626Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Windows_NT x64 10.0.19045

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around printing non-ASCII characters. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.

Odud, 2026-09-14:
Works fine now on MicroPico 4.3.4, The sad thing is I can't remember why I needed it work...

closed 2026-09-14 by Odud