paulober/MicroPico #266: REPL not reconnecting after machine.USBDevice is reactivated

HLammers, 2024-12-19:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
Install requirements for [midi_example.py](https://github.com/micropython/micropython-lib/blob/master/micropython/usb/examples/device/midi_example.py) from micropython-lib and run it (I had to rename it to main.py).

## What happens?
The REPL gets disconnected at the moment `machine.USBDevice.active(False)` is called (as step in `_Device.init` in [core.py](https://github.com/micropython/micropython-lib/blob/master/micropython/usb/usb-device/usb/device/core.py), but doesn’t reconnect when `machine.USBDevice.active(True)` is called (as a next step in `_Device.init`).

## What were you expecting to happen?
I’d expect the REPL to reconnect.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Extension Host and copy output concerning pico-w-go or micropico)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
2024-12-19 14:22:43.372 [info] [Stubs] Installed stubs are already up to date!
2024-12-19 14:33:01.427 [info] [Activator] Connection to board successfully established
2024-12-19 14:34:42.452 [info] [Activator] Connection to board was closed.
2024-12-19 14:34:43.472 [info] [Activator] Connection to board successfully established
2024-12-19 14:34:50.794 [info] [Activator] Connection to board was closed.
2024-12-19 14:35:25.982 [info] [Activator] Connection to board successfully established
2024-12-19 14:35:45.217 [info] [Activator] Connection to board was closed.
2024-12-19 14:35:46.251 [info] [Activator] Connection to board successfully established
2024-12-19 14:35:49.967 [info] [Activator] Connection to board was closed.
2024-12-19 14:35:50.984 [info] [Activator] Connection to board successfully established
2024-12-19 14:36:43.446 [info] [Activator] Connection to board was closed.
2024-12-19 14:36:44.481 [info] [Activator] Connection to board successfully established
2024-12-19 14:36:50.254 [info] [Activator] Connection to board was closed.
2024-12-19 14:36:51.283 [info] [Activator] Connection to board successfully established

## Any other comments?

## Which version of MicroPico are you using?
v4.1.0

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.96.1 (user setup)
Commit: 42b266171e51a016313f47d0c48aca9295b9cbb2
Date: 2024-12-17T17:50:05.206Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Windows_NT x64 10.0.26100
```

masterkain, 2025-05-26:
same issue using

```
import usb.device
from usb.device.keyboard import KeyboardInterface

    k = KeyboardInterface()
    usb.device.get().init(
        k,
        builtin_driver=True
    )
```

paulober, 2026-09-14:
[MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) changed a lot around connecting and reconnecting boards. Could you check if this still happens with 4.4.0? If there's no reply within 30 days, this issue will be closed automatically.