paulober/MicroPico #269: Unplugging device while running causes status bar "Stop" button to stay as "Stop" forever.

DeflateAwning, 2024-12-22:
## What are the steps to reproduce this issue?
1. Run a program with the run button from the status bar (bottom bar in VS Code).
2. Unplug the board while it's running.
3. See that the Stop button remains.
4. Try clicking the Stop button.
5. Plug device back in.
6. See that the button stays as the "Stop" button.
7. Try clicking the Stop button.

Nothing you do can make the Stop button turn back into a Start button, even though the device has been reconnected.

## What were you expecting to happen?

I'd expect that the Stop button disappears when the board is disconnected, and turns back into a Start button upon reconnection.

## Which version of MicroPico are you using?

VS Code release version: `2024-11-17, 17:16:36`

realfun, 2024-12-31:
Having same issue: Version: 1.96.2 (Universal), plugin version: 4.1.0; the START/STOP button on status bar remains as STOP until I restart VS code.

closed 2025-01-17 by paulober

DeflateAwning, 2025-02-01:
Awesome fix, thank you! Fix works for me!

This extension is so more pleasant to use now, especially when using it to hack together tests for a failure-prone PCB in development :)

commit c3c88cd8 (2025-01-17): Fix #269, Frozen 'Stop' button after unplugging
PR #274 (2025-01-17): Patch v4.2.0
### Changed

- Updated dependencies
- Configure project command has been renamed to `Initialize MicroPico project`
- Fixed #269, Frozen 'Stop' button after running unplugging
- Refined the disconnect behavior