paulober/MicroPico #279: Sync folders from outside the project root

jkozniewski, 2025-01-23:
# Feature Request 🚀

It would be great to have an option to sync some arbitrary folder specified by an absolute path (or relative but above the project root so "../")
The reason for such feature are projects that are using some shared resource - like a library code or some shared config that's get deployed on separate boards. If there is any solution for having a shared code between the projects that's possible in current config even without such feature I'd be grateful for pointing it out.
Thanks !

jkozniewski, 2025-01-23:
Ok.. I guess I have found a solution - at least working on mac os - using symbolic links.
So in case someone has same need to share some code between separate micropico projects:

1. create a symlink `ln -s /path/to/real_folder /path/to/link`
2. in `settings.json` of the project add in `python.analysis.extraPaths` path to the lib - if it's just above the rot of the project it might be relative so `"../lib"`

Tested and both - code completion and uploading to the board respects the `lib` folder outside of the project root :)
Also fortunately git is smart enough to handle the symlinks properly - so the code is not duplicated in the repo.

paulober, 2026-09-14:
Thanks for the suggestion! I'm keeping sync folders inside the project on purpose, and the symlink approach you found above covers sharing code between projects, so I'm closing this.

closed 2026-09-14 by paulober