paulober/MicroPico #282: Some bytes printed to terminal have extra bytes

StevenCollins, 2025-01-31:
When printing some bytes to the terminal, unexpected bytes appear.

## What are the steps to reproduce this issue?
1. Open a connection to a Raspberry Pi Pico W with RP2040
2. Type `b'\xAA'` in the terminal

## What happens?
`b'\xc2\xaa'` is printed to the console

## What were you expecting to happen?
`b'\xaa'` to be printed to the console

## Any logs, error output, etc?
No

## Any other comments?
This functions as expected in a normal Python environment as well as using Thonny with the same Pi Pico.
It works as expected with printable bytes. ex. `b'\x7a'` prints `z`.
I'm not positive, but seems that the byte is being interpreted as Unicode, then the full UTF-8 literal is printed. As in, `\xAA` is interpreted as `U+00AA` which has the UTF-8 literal `\xc2\xaa`.
This may be related to https://github.com/paulober/MicroPico/issues/265

## Which version of MicroPico are you using?
4.2.1

## Support info

Copy this from the `Help -> Info/About -> Copy` (`Code -> About Visual Studio Code -> Copy` on macOS) option in Visual Studio Code:

```
Version: 1.96.4 (user setup)
Commit: cd4ee3b1c348a13bafd8f9ad8060705f6d4b9cba
Date: 2025-01-16T00:16:19.038Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Windows_NT x64 10.0.26100
```

dmoisset, 2025-02-02:
I can confirm the same situation in Linux (also using MicroPico 4.2.1):

```
Version: 1.96.4
Commit: cd4ee3b1c348a13bafd8f9ad8060705f6d4b9cba
Date: 2025-01-16T00:16:19.038Z
Electron: 32.2.6
ElectronBuildId: 10629634
Chromium: 128.0.6613.186
Node.js: 20.18.1
V8: 12.8.374.38-electron.0
OS: Linux x64 6.8.0-52-generic snap
```

This is quite inconvenient because operations like `spi.write('\xff')` send the wrong number of bytes.

BobStein, 2025-06-25:
I discovered this today. Rather frustrating in a language for embedded systems where lots of interfaces use bytes, such as I2C.  

Not sure this is the same issue as the [Printing Emojis garbage](https://github.com/paulober/MicroPico/issues/286) issue. This bug has nothing to do with display, it's a failure of the [byte-string literal](https://docs.python.org/3/reference/lexical_analysis.html#string-and-bytes-literals) syntax in the MicroPico REPL.

Another way to demonstrate it. Using the MicroPico terminal window:

```
>>> len(b'\xFF')
2
```

But using the MicroPython REPL (that is, connecting to the Pico using PuTTY on the COM port):

```
>>> len(b'\xFF')
1
```

Apparently: a byte-string literal is interpreted as a UTF-8 encoding. With the MicroPico REPL:

```
>>> b'\xFF' == '\xFF'.encode('utf-8')
True
```

That should not be true.

Josverl, 2026-01-14:
Just fyi, 
There were some Unicode issues that needed fixing in Mpremote as well.

Might provide an alternative way to inspect.

https://github.com/micropython/micropython/pull/18670

paulober, 2026-09-14:
Fixed in [MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0): backslashes in vREPL expressions are no longer corrupted, so `b'\xAA'` prints `b'\xaa'`. Thanks for reporting!

closed 2026-09-14 by paulober