paulober/MicroPico #286: Printing emojis shows as garbage in console

DeflateAwning, 2025-02-21:
> Please provide a clear and concise description of the bug along with logs

## What are the steps to reproduce this issue?
1. Click run on a program like: `print("❌ Failure")`
2. See that the printed message appears like: `��� Failure`

## What were you expecting to happen?
It should have printed this emoji: https://emojipedia.org/cross-mark

## Which version of MicroPico are you using?
4.2.1

## Support info

VS Code on Linux

Josverl, 2025-02-21:
This appears to related to the configuration of the terminal with regards to UTF-8 in order to support Unicode.

harvey2953, 2025-02-26:
if print("中文"), I have the same issue

DeflateAwning, 2025-02-26:
Seems like it should probably be solved in this extension's terminal configuration.

Josverl, 2025-02-26:
There seems to be very little configuration for the (External)Terminal in vscode, all terminals support Unicode without config.
So I expect it is the terminal Add in that needs attention.

DeflateAwning, 2025-03-15:
Yeah VS Code's terminal definitely supports unicode by default. Am very curious what this extension did to cause an issue with that

DeflateAwning, 2025-04-21:
Feels like extremely bad practice (and outright unnecessary) to parse characters like that. There are encode and decode functions in every language for that purpose. 

I haven't looked too deeply into exactly what's going on there, but seeing that type of code written out feels like a Problem X-Y, where the real issue is a step away from the function written there. 

Potential fix:

```typescript
function decodePythonStyleEscapes(str: string): string {
  return JSON.parse(`"${str.replace(/"/g, '\\"')}"`);
}

// Then in your writeEncodedBufferToFile, just do:
const processed = decodePythonStyleEscapes(escaped);
const binaryData = Buffer.from(processed, "utf-8");
await fileHandle.write(binaryData);

```

paulober, 2026-09-14:
Fixed in [MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0): printing emojis and other UTF-8 characters works correctly (the decoding fix is included). Thanks for reporting!

closed 2026-09-14 by paulober