paulober/MicroPico #294: After KeyboardInterrupt, all commands give: Error executing commandException occured [sic]

BobStein, 2025-06-12:
The getting-started program at the [home page](https://github.com/paulober/MicroPico) works fine if I paste it using PuTTY on the COM port (using Ctrl+E paste mode). After checking that the LED blinks, I type Ctrl+C to end the loop. Then commands work as expected (screenshots below)

But if I put that code in a file in VSCode, and run it via the MicroPico command "Run current file on Pico", and THEN type Ctrl+C, execution finishes, but no commands work after that. No matter what I type I get a long (5ish seconds) delay and this error:

 `Error executing commandException occured`

Working via PuTTY:

[image]

Failing via MicroPico:

[image]

In my immediate application, I'm trying to come up with a way to abort a printing-status loop from the terminal in VSCode. KeyboardInterrupt would have helped. But then I have to reset the Pico to continue. I tried a workaround making a kbhit() style polling function, but it never detects any keystrokes, they all seem to be intercepted by the terminal UX and never sent to the Pico.

MicroPython v1.25.0

MicroPico 4.2.2

VSCode:
Version: 1.100.3 (user setup)
Commit: 258e40fedc6cb8edf399a463ce3a9d32e7e1f6f3
Date: 2025-06-02T13:30:54.273Z
Electron: 34.5.1
ElectronBuildId: 11369351
Chromium: 132.0.6834.210
Node.js: 20.19.0
V8: 13.2.152.41-electron.0
OS: Windows_NT x64 10.0.26100

idrisibrahimerten, 2025-06-30:
Click on the “Pico Connected” text at the bottom left to disconnect. Then click it again to reconnect, and select “Reset”. After that, click “Run” to execute your desired operation—it should work properly.

paulober, 2026-09-14:
Fixed in [MicroPico 4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0): commands keep working after stopping a program, and Stop also works again when a program ignores the first interrupt. If you still run into this, comment here and I'll reopen it. Thanks for reporting!

closed 2026-09-14 by paulober